CharaChorder/CCOS-firmware #3: Defect: chording a word after arpeggiated punctuation deletes the space after the punctuation

jdestgermain, 2023-06-22:
Firmware: v1.1.1, CharaChorder One

Steps to reproduce:
1. Chord a word
2. Arpeggiately add punctuation, such as a period
3. Chord the first word of the next sentence

Expected outcome:
- It should add the chord and keep the space between the punctuation and the chord

Actual outcome:
- It deletes the space 

Example: 
https://github.com/CharaChorder/CCOS-firmware/assets/11359044/c7b09506-2e1d-470d-a5d9-5278a9273c5b

mike-mosher, 2023-08-20:
any update on this one?  it makes it very hard to chord.

squeakyboots, 2023-08-21:
@mike-mosher the priority board and what is being working on can be seen [here](https://github.com/orgs/CharaChorder/projects/1).

mike-mosher, 2023-09-02:
Just checked the board and noticed this bug as a priority.  I'm excited to hear that.  This honestly is the only issue I face day to day using the CC, and it is super disruptive to the work flow.  Thanks!