CharaChorder/CCOS-firmware #7: Defect: while in spurring, punctuation is removing the last letter of the last word

jdestgermain, 2023-06-23:
Firmware: v1.1.1, CharaChorder One

Steps to create:
1. Enter spurring mode
2. Chord a word
3. Add punctuation

Expected output:
- It should add the punctuation and a space

Actual output: 
- It deletes the last letter of the word

References:
[image]
https://discord.com/channels/861730583092658206/1096420269687578635

squeakyboots, 2023-07-03:
I also get this behavior when doing regular chording. I confirmed it persists in v1.1.1. Per https://discord.com/channels/861730583092658206/938239528454475787/1102576292202819725 there should be a fix in one of the betas..?

It seems easier to recreate it by increasing the arpegiate timeout in the settings here:  CharaChorder > Chording > Arpeggiate [ >O<N/off=ON || >T<imeout=**600ms** ]

The repro scenario: do a chord, then character entry a word quickly and use punctuation such as "." or ",". The punctuation will remove the last letter of the character entry word.

@jdestgermain should a separate issue be raised for this behavior in non-spurring mode, too? Or are they considered the same thing?

squeakyboots, 2023-07-06:
@Raymo111 maybe..? I'm not sure who is the most relevant resource on this.

Raymo111, 2023-07-06:
@strontium5 is the key CCOS guy, I'm not sure if this is the same issue or not.

Theaninova, 2025-06-30:
Closing some older issues, if you think this is still relevant please reopen.

closed 2025-06-30 by Theaninova