CharaChorder/CCOS-firmware #14: Defect: Punctuation will remove trailing character in some scenarios, U + P chords "up " despite not being a chord I've maintained

squeakyboots, 2023-07-06:
**Describe the bug**
When I chord certain words then character entry and do a punctuation the character entry will cut the tail off the last word typed.

I think there may be more than one scenario where this is happening, but one I've pinned down involves the chord U + P for the word "up". I do not have U + P in my chord library but it will still trigger "up " as if chorded. If I then repeat U + P together it will not trigger the chord again, but if I press "." it will cut off the last letter and allow me to use the chord U + P again.

A point of clarification: I do not want U + P to chord "up" and do not have that chord in my library, yet it still seems to exist when using the device.

**To Reproduce**
[3m8s video](https://drive.google.com/file/d/1hML6AeMTtwJ0PgJNwzwlRhENGvSjG1H_/view?usp=drivesdk)
1. Remove U + P as a chord from your library.
2. Chord "up " using the chord U + P that has been removed from your library repeatedly. Output should look like this: up uppuupup
3. Press ".". 
4. Output will go from "up uppuupup" to "up uppuupu." (note the last "p" was dropped and replaced with a ".")

**Expected behavior**
1. U + P doesn't chord "up" if I don't have that in my chord library.
2. "." doesn't cut off the last letter of my last chord.

**Please complete the following information:**
 - OS: Microsoft Windows 10 Pro, 10.0.19045 Build 19045
 - CharaChorder Device: CC1
 - CCOS Version: 1.1.1

squeakyboots, 2023-07-13:
Even after clearing all chords U+P has the same behavior as described above: [36s video](https://drive.google.com/file/d/1NA-6q-EYDy7FOW3xtHk3iMMHp5H-F15I/view?usp=drivesdk)

squeakyboots, 2023-07-18:
This issue persists after a factory reset: [5m14s video](https://drive.google.com/file/d/1eSZHQvl0ejT4oAPkRuCkBiW803lRFhtn/view?usp=drivesdk)

rileykeen, 2023-07-19:
**Bug 1:** I am able to delete the chord "up" without any issue, and do not believe the bug you have identified here is attached to any specific memory location, but rather is generally applicable to any instance that the chord detection algorithm is triggered regardless of whether or not the user has an output assigned to the chord inputs triggered.

Steps to reproduce:
1. Chord any inputs which are not assigned to a chord output on your device. (Ex: "kfuoetns")
2. Immediately follow this false chord with a punctuation mark tied to the auto-append functionality (.,?; etc)
3. The auto-append functionality will erroneously trigger (EX: "kfuoetn. ")

**Bug 2:** I have also found through testing this, an issue which occurs when chording unmapped inputs immediately following an auto-append punctuation mark:

Steps to reproduce:
1. Chord anything immediately followed by a punctuation mark tied to the auto-append functionality (.,?; etc)
2. After the auto-appended punctuation is applied, chord any inputs which are not assigned to a chord output on your device (Ex: "kfuoetns")
3. A spacebar will be appended to them as if they were a chord (Ex: "kfuoetns ")

rileykeen, 2023-07-19:
My pleasure. That sounds great, I agree :)