CharaChorder/CCOS-firmware #18: Enhancement: Media control keys

rccarlson, 2023-07-20:
**Is your feature request related to a problem? Please describe.**
Would like to be able to control media via the CC. Inputs such as volume control or media skip/pause/play.

**Describe the solution you'd like**
Added key codes, configurable from charachorder-config.com, or more preferably Dot I/O.

**Describe alternatives you've considered**
No alternative that I'm aware of. This is something that must be supported by the keyboard according to my research. Otherwise, simply out of luck.

jytou, 2023-10-01:
To me, this is more a bug than an enhancement, though… these are quite standard these days, and not being able to manipulate volume and play/pause music is a huge regression for me compared to using the keyboard without the CCX.

xslendix, 2024-04-28:
NOTE: The way keyboards have those functions is not as keycodes, but as system control functions which are a different descriptor. This would require a new USB descriptor to be implemented into CCOS and special code to act on that for the keys

wkumari, 2024-05-10:
Yes, but would require the current descriptor to be modified, but Charachorder X already advertises multiple descriptors. 
AFAICT, the only "special code" needed would be to simply transparently pass through any "keypresses" (events) which it doesn't recognize. 

The current CC X descriptors include: 
```
0x05, 0x01,  0x09, 0x02 [...]     // Generic Desktop Ctrls,  Usage Mouse - Usages: Pointer, X, Y, Wheel, AC Pan.
0x05, 0x01,   0x09, 0x06 [...]     // Generic Desktop Ctrls, Usage Keyboard - Usages: Kbrd/Keypad, LEDs, Kbrd/Keypad 
```
        
I think that all that would need to be added is:
0x05, 0x0C,  0x09, 0x01,   0xA1, 0x01  // Usage Page: Consumer,  Usage: Consumer Control, Collection: Application [...]
The full additional descriptor: `05  0c  09  01  a1  01  85  04  19  01  2a  a0  02  15 01  26  a0  02  95  01  75  10  81  00  c0
`
The current CC X descriptor:
```
  05  01  09  02  a1  01  85  02  09  01  a1  00  05  09  19  01
  29  05  15  00  25  01  95  05  75  01  81  02  95  01  75  03
  81  01  05  01  09  30  09  31  15  81  25  7f  95  02  75  08
  81  06  09  38  15  81  25  7f  95  01  75  08  81  06  05  0c
  0a  38  02  15  81  25  7f  95  01  75  08  81  06  c0  c0  05
  01  09  06  a1  01  85  01  05  07  19  e0  29  e7  15  00  25
  01  75  01  95  08  81  02  75  08  95  01  81  01  05  08  19
  01  29  05  75  01  95  05  91  02  75  01  95  03  91  01  05
  07  19  00  2a  ff  00  15  00  26  ff  00  75  08  95  0c  81
  00  c0
```
(146 bytes)

The modified one:
```
  05  01  09  02  a1  01  85  02  09  01  a1  00  05  09  19  01
  29  05  15  00  25  01  95  05  75  01  81  02  95  01  75  03
  81  01  05  01  09  30  09  31  15  81  25  7f  95  02  75  08
  81  06  09  38  15  81  25  7f  95  01  75  08  81  06  05  0c
  0a  38  02  15  81  25  7f  95  01  75  08  81  06  c0  05  0c
  09  01  a1  01  85  04  19  01  2a  a0  02  15  01  26  a0  02
  95  01  75  10  81  00  c0  c0  05  01  09  06  a1  01  85  01
  05  07  19  e0  29  e7  15  00  25  01  75  01  95  08  81  02
  75  08  95  01  81  01  05  08  19  01  29  05  75  01  95  05
  91  02  75  01  95  03  91  01  05  07  19  00  2a  ff  00  15
  00  26  ff  00  75  08  95  0c  81  00  c0
```
(171 bytes)

TeamDman, 2025-01-21:
I have media pause/play bound on my Logitech g502 and would love for it to work on the CC2. The charachorder.io site let's me bind the media play pause button but pressing it does nothing :(

xslendix, 2025-01-22:
@TeamDman The reason is that operating systems don't use these keys for anything. All media buttons on keyboards, at least modern ones, are done using the descriptor @wkumari mentioned.

TeamDman, 2025-02-24:
maybe also the cause of the "menu" key not working on windows

frankbenoitbosch, 2025-03-02:
As workaround on windows for CC1:

Use Autohotkey

````
#SingleInstance force
F20::Send "{Volume_Up}"
F21::Send "{Volume_Down}"
F22::Send "{Volume_Mute}"
F23::Send "<#!k"
````

The last one for the Microphone mute toggle. Store in a file with .ahk extension, and put a shortcut into the autostart folder.

frankbenoitbosch, 2025-03-02:
@TeamDman the windows menu key works for me with CC1, but it is the "Keyboard Application" key that shall be used.

TeamDman, 2025-03-11:
Woah, thank you @frankbenoitbosch, that works for me!

closed 2025-06-30 by Theaninova