CharaChorder/CCOS-firmware #50: Defect: Punctuation chords still work in character only mode

rileykeen, 2023-10-03:
**To Reproduce**
Steps to reproduce the behavior:
1. Use exclamation chord while character only mode is activated

**Expected behavior**
A clear and concise description of what you expected to happen.
Chords should not work in this mode, however some do

**Screenshots**
If applicable, add screenshots to help explain your problem.
https://discord.com/channels/861730583092658206/1158409114658029608/1158409114658029608

bjacobgordon, 2023-11-27:
@rileykeen Is there any timeline for this yet?

mattswartscc, 2023-11-27:
There is a category of 'hardcoded' chords which should not be disabled when character only mode is activated. These are primarily the GTM and Impulse chords. However, on the CC1, there are about 15 additional 'hardcoded' chords which do not fall into this necessary category. We will migrate these to starter functional chordmaps, which can be re-added with the Serial API command of RST FUNC in a serial terminal. This is a relatively simple modification that we will try to get into the next release. We are currently testing an alpha this week.

bjacobgordon, 2024-01-25:
@mattswartscc Thanks for the breakdown!

Still dealing with this issue on a daily basis, what's the status for this fix.

bjacobgordon, 2024-04-22:
Working for me so far!