CharaChorder/CCOS-firmware #67: Defect: QMK Auto Shift breaks chording

bhansconnect, 2024-01-09:
**Describe the bug**
If a QMK keyboard with the [auto shift feature](https://docusaurus.qmk.fm/feature_auto_shift/) enabled is connected to a CharaChorder X, chording will not work at all. Even the `Esc + G` chord will fail.

The keyboard will still be able to type, but upon letting go of all keys, the last pressed key will repeat forever. This specific part is the same defect as described in #32, just caused by auto shift instead of tap dance.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable AutoShift on a QMK keyboard
2. Flash the device
3. Plug it into a CharaChorder x
4. Attempt to chord anything

**Expected behavior**
If I press a chord like `bc`, it should output `because` instead of just outputting `bc`. Also, if I tap `c`, it should just output a single `c` instead of repeating the character forever.

**Please complete the following information:**
 - OS: MacOS 14.2.1
 - CharaChorder Device: CharaChorder X
 - CCOS Version: 1.1.3

**Additional context**
Used a Corne crkbd with QMK. Should theoretically affect any QMK keyboard with autoshift enabled.

bhansconnect, 2024-01-12:
Found part of a fix but not a full fix:

`#define TAP_CODE_DELAY <num>`. Apparently when using auto shift, qmk will default to 0 delay when pressing a key that isn't shifted. As a result, the key press and release events will be fired one after another. I think that CCOS misses the release event and just gets stuck with the key press down.

Sadly, this is not a full fix for my issue above. I still have no chording. My guess is that the artificial key presses are generate such that the charachorder doesn't register all presses at the same time. Instead it sees them as only slightly overlapping or something like that. As a result, chording never triggers.

Theaninova, 2024-01-24:
I think what's going on

> When you press a key, a timer starts, and if you have not released the key after the `AUTO_SHIFT_TIMEOUT` period, then a shifted version of the key is emitted. If the time is less than the `AUTO_SHIFT_TIMEOUT` time, or you press another key, then the normal state is emitted.

Auto-shift does not output anything until you release the key. It's critical for ccos to be able to get precise timings, simply having a "tap" is not enough since events only get send after each release which looks like sequential taps to ccos.

So I don't think this is gonna ever work unfortunately, and I'd recommend having a shift key either way for chording.