CharaChorder/CCOS-firmware #88: Enhancement: More layers (eg: to support OSX and PC)

graemeg, 2024-07-26:
**Is your feature request related to a problem? Please describe.**
My daily driver keyboard is still my Ergodox, because CC1 doesn't fit my use case perfectly yet. I use MacOS for work, and a PC (FreeBSD) for personal use. More layers will help me keep muscle memory without having to think of different keys, based on my OS i'm using at the time. 

**Describe the solution you'd like**
Can we get more layers? For example, my Ergodox keyboard supports up to 32 layers. I use 8 in my standard layout, seen here: https://configure.zsa.io/ergodox-ez/layouts/APGaG/latest/0

I know with my CC1, I achieve most of that with the standard 3 layers, but I can see situations where you might want more. 

To give one example, when I'm working, on my work (MacOS) laptop [vs my home FreeBSD desktop], I toggle the OSX layer, so it switches my Ctrl & Cmd keys. So Ctrl+S or Cmd+S is always the same action and muscle memory - I simply had to toggle the default layer at the beginning and end of day.

Another example is my "Layouts" layer, which I use to manage windows on my desktop. So I can easily position windows with keyboard shortcuts.

Other users would find this useful too, as discussed on Charachorder's discord here: https://discord.com/channels/861730583092658206/1260643008353665045/1260643008353665045

**Describe alternatives you've considered**
My CC1 has very limited layers. Only 3.

**Additional context**
See my Ergodox layout in Oryx, linked above.

graemeg, 2024-07-26:
Actually I'm referring to both, as I use both daily. Toggle layers (TO) and Momentary-Toggle layers (MO). I use TO for example for switching between MacOS and PC "mode" (where it switches the Cmd & Ctrl keys). I do this at the beginning and end of my work days. I also use MO layers throughout the day for accessing Arrows or Symbols without having to move my fingers off the home row. This latter one is slightly less of an issue on the CC1, due to 5 directional keys, but I think they could still be useful if they were available. I could see myself doing similar to my Ergodox layout. Use a thumb to temporarily switch (MO) to the Symbol layer and the use 3 fingers on each hand has access to all symbols (without needing to type Shift). This would be excellent for programming, and much less hand movement than I currently have to do on the CC1.

instantepiphany, 2024-11-17:
I'd really appreciate this too. 

With my moonlander, I can toggle my "gaming layer" which mostly turns the smart features off (so keys act more like a normal keyboard, which is what games expect).

On the CC2, I tried to set it up for games, and found the smaller sticks (default to arrow keys and mouse movement) are very hard to get diagonal outputs on, where the other sticks are much easier to get diagonal outputs on. 

I like the default CC2 layout, it is pretty well designed and I want to keep it. 

So if I could enable a gaming layer that put WASD for example on the index or middle finger stick, and maybe move some modifiers around, that would be fantastic. 

If I want to do that at the moment, I have to backup my "normal" layout, change the keys to the gaming setup, and then later restore the backup - which means if a game requires any text input, I can't do it unless I also learn to type on this gaming layout. 

On my moonlander, with one key I can switch between the layouts.

@jdestgermain what @graemeg and I are missing (and other users might appreciate too) is essentially [the layer feature from QMK](https://docs.qmk.fm/feature_layers#switching-and-toggling-layers).

I'm hoping that the secret CC2 feature announced soon is QMK firmware support (or equivalent features) as that would enable users to make use of a pre-existing community and ecosystem for tailoring devices to their needs.