CharaChorder/CCOS-firmware #98: Enhancement: Open Sourcing CCOS

nyabinary, 2024-09-02:
Most enthusiast-grade keyboard firmware is [open source and supported by various FOSS projects](https://www.reddit.com/r/ErgoMechKeyboards/comments/17fr7te/tmk_vs_qmk_vs_kmk_vs_zmk/). While it’s understandable that CharaChorder needed custom firmware for the CC1, Master Forge, and other devices to support their unique chording capabilities, I believe open-sourcing this firmware could significantly accelerate development and broaden its user base. Some potential users may avoid the product simply because the firmware is proprietary. Open-sourcing aligns perfectly with the Forge principle of making the keyboard truly “yours,” as it would allow users to modify the source code and customize the functionality to their liking. Moreover, by giving users even more control and customizability, CharaChorder could foster a stronger, more engaged community. It could also drive innovation through collective contributions, enhance security and stability through peer review, and ultimately create a more dynamic and trusted product that evolves rapidly in response to user needs. 

NOTE: I'm obviously pretty biased here as a Linux user and FOSS contributor :3, but I would love to see the CCOS be open sourced.

whited, 2024-09-19:
I have pledged for the Forge KickStarter, and I for one would LOVE to be able to tinker with and contribute to the firmware.  I hope this idea is seriously considered!

Hamcha, 2024-10-07:
Couldn't help but notice that at least the CharaChorder Engine firmware contains symbols for [SPIMemory](https://github.com/Marzogh/SPIMemory), a GPL-3.0 licensed library, which requires all derivatives to be licensed under the same license and all binaries provided with accompanying source code.

nyabinary, 2024-12-05:
> Couldn't help but notice that at least the CharaChorder Engine firmware contains symbols for [SPIMemory](https://github.com/Marzogh/SPIMemory), a GPL-3.0 licensed library, which requires all derivatives to be licensed under the same license and all binaries provided with accompanying source code.

What does this mean in theory?

ls-1N, 2025-01-19:
Oh, I feel a bit misled 😂 - I assumed that the mere existence of this repo meant that CCOS _is_ open-source. I didn't actually check the repo contents (and lack of a license) showing that in this case Github is simply used as a means to host the binaries(?) and basically a issue ticket system.

> What does this mean in theory?
My uninformed guess would be that, legally, CCOS is required to also be open source?