CharaChorder/CCOS-firmware #104: Defect: Mirror Key Sustains Ctrl If Released First

mtevebaugh, 2024-12-27:
**Describe the bug**
Mirror key + Modifier sustains modifier if mirror is released first

**To Reproduce**
Steps to reproduce the behavior:
1. Press Left Mirror + Ctrl + Other i.e. Ctrl + D for copy down in Excel
2. Release Mirror **before** Ctrl
3. Ctrl is sustained pressed
4. Press Mirror + Ctrl again and release Ctrl first to clear

**Expected behavior**
Releasing Ctrl performs Ctrl release even if performed mirrored in whatever sequence

**Screenshots**
NA

**Please complete the following information:**
 - OS: Edition	Windows 11 Pro
Version	23H2
Installed on	‎11/‎10/‎2023
OS build	22631.4602
Experience	Windows Feature Experience Pack 1000.22700.1055.0
 
 - CharaChorder Device: CHARACHORDER TWO S3
 - CCOS Version: 2.02

**Additional context**
Happens with both mirror keys. I have my alt and mirror in default layout and I dont want to remap to test, but this may also be true for other modifier keys like Alt and shift.

mtevebaugh, 2024-12-27:
Ah that makes sense. Yeah I imagine making this a toggle feature is not insignificant. Now that I understand what's happening, it isn't a huge deal. I can try to think about how I would use it as a feature.

Maybe an alternative could be a special sustain key in ccos. Not assigned in the default layout, probably. But if you press that at the same time as any other single key it could sustain it until pressed together again or escaped or something. Not sure if that's more complicated and maybe it only works with modifiers. Shrug. Seems like there are ways to keep the functionality since it definitely seems useful.

Vaelatern, 2025-03-31:
This is great in the mouse. The mouse is a modifier too it seems, and you can "lock" the mouse running away. It's fun, the way to stop it is hit ambi-throw on the OTHER puck, and drive the mouse in the opposite direction. This ends up bringing the mouse speed back to zero.

Only hitting ambi-throw on the other puck works to actually zero the mouse speed.