CharaChorder/CCOS-firmware #119: A replug is required to make manager or impulse chords work.

Theaninova, 2025-06-30:
Added a chord and saved it in the manager: https://charachorder.io/
chord: ar
output: are

Chording: ar
It only typed the characters, the chord didn't work.

Deleting the chord and saving, removed it from the manager. (expected)
But syncing the manager brought it back. (unexpected)

Chording: ar
The chord didn't work.

Replugging.
Chording: ar
It works. are are are

The same thing happens with the impulse prompt.
output: close
input: clo
The chord doesn't work.
Syncing the manager shows the chord.
I didn't try removing the chord and syncing again.
Replugging.
The chord works: close close close

duianto, 2025-07-02:
This seems to have been introduced in 2.2.0-beta.2.
> CCOS Releases APP — Thu 19 Jun 2025 ⁠https://discord.com/channels/861730583092658206/950760334313029672/1385310720911999026
>  one_m0 2.2.0-beta.2 two_s3 2.2.0-beta.2 m4gr_s3 2.2.0-beta.2 lite_s2 2.2.0-beta.2 m4g_s3 2.2.0-beta.2

> Theaninova [SAVE]  — Thu 19 Jun 2025
> This one improves flash performance, primarily noticeable with chord importing.

### Issue reproduced
- M4G 2.2.0-beta.2 and 2.2.0-beta.3

### Expected behavior. New impulse or manager chords work immediately
- M4G 2.2.0-beta.1
- CC2 2.2.0-beta.3
- CC1 2.2.0-beta.3

neighthan, 2025-07-16:
I'm having a similar issue with impulse chords showing up in the manager but not working. It doesn't happen every time; some chords I make will work but not all. Replugging doesn't seem to be a consistent solution, though, and sometimes my chords are just deleted and/or a weird chord is added whose input is ACTION_MAX repeated multiple times and whose output is a very long string of yyyyyyyyyyyyyyyyyyyyyy but with umlauts. I'm on 2.2.0-beta.5 now on a m4g.

Also, chords with two letters consistently don't work.

DV86, 2025-07-16:
https://discord.com/channels/861730583092658206/1378636098363850762

I thought I was the only one to have this issue; now I can't record any chord as it would mess it all up.

duianto, 2025-07-16:
> I'm having a similar issue with impulse chords showing up in the manager but not working. It doesn't happen every time; some chords I make will work but not all. Replugging doesn't seem to be a [...]

@neighthan 
The action_max issue occurred before 2.2.0-beta.2, so it's unrelated to this replug to make chords work issue.

Could you open a new action_max issue.

DV86, 2025-07-16:
I still get the issue with some special characters..